olliw42/mLRS-docu #14: Add "wiring" graphics for most common setups

olliw42, 2023-03-29:
one probably could add the graphics of rcgroups post https://www.rcgroups.com/forums/showpost.php?p=48821735&postcount=332 to proper places in the docs (they are linked to in the mLRS main README page, but this is maybe not prominent enough to be useful)

the pics need some update too as some info has slightly changed

triggerd by discussion https://www.rcgroups.com/forums/showpost.php?p=50757843&postcount=1555, https://www.rcgroups.com/forums/showpost.php?p=50758181&postcount=1556

jlpoltrack, 2023-03-29:
As opposed to these graphics, what do you think about a table showing the connections that need to be made for all the hardware on both the Tx and Rx side?  I found these block diagrams a bit too high level :)

Example:

[image]

olliw42, 2023-04-02:
I frankly have no opinion here. To me it always appears like this one user prefers this and this other user prefers that. From pure statistics it seems to me that wire diagrams are more frequently found (JB even draws them in realtime in his live streams when he wants to show a wiring). That's all I can say LOL.

jlpoltrack, 2023-04-02:
Added an example wiring diagram for the E5 Mini as Tx in this branch: https://github.com/olliw42/mLRS-docu/tree/Seeed-E5-Wiring.

I think it makes sense to try and complete these for each hardware page.  Can worry about the DIY hardware later :)

olliw42, 2023-04-03:
N I C E !!

shouldn't one maybe also add the green led?

for a short moment I was confused about the top grey line as it seemed to have no label, until I realized that's serial rx,tx. Maybe just some centering, or indeed writing serial Rx, serial Tx, could help

since there is space, one could consider also adding a S-diode suggestion like e.g. BAT42

but I'm really nit-picking now, it's a great and very helpfull drawing

olliw42, 2023-07-09:
I think the intended graphics are now in the respective setup descriptions.

It probably could be useful - at some point in time - to have a high-level intro-like page with brief descriptions on the various possibilities possible with mLRS. But I think for now this here can be closed.

Pl do not hesitate to reopen if deemed appropriate. :)

closed 2023-07-09 by olliw42