olliw42/mLRS-docu #50: Add Tx Snd RadioStat suggestion to SiK page

jlpoltrack, 2023-07-09:
To have the RSSI of both radios, this parameter should be set (SiK does this).

olliw42, 2023-07-09:
that's a most interesting comment, indeed

with regards to SiK specifically (and SBus), I think that's a good suggestion. It raises the question however if or why not it should be added to the other pages.

The RSSI topic is non trival. First as regards its sources and also as regards its value range. I would hope that there could be some canonicalization, by e.g. new MAVLink messages as suggested here https://github.com/mavlink/mavlink/issues/1921 but the resistance to move is high, ArduPilot is largely driven by money today (partners, devs, companies) as it seems. So, we have to live with how it is.

AFAIK there can be/are 3 MAVLink sources of rssi from the flight controller, i.e., the receiver. As regards the rssi from the tx module, there is only one MAVLink source, the RadioStat message. But - except for SiK/SBus - there is also the CRSF data.

Maybe a RSSI page giving an overview of all options and approaches would be the best. But for the moment, adding it to Sik is probably a good idea, given that most people will be just confused otherwise.

jlpoltrack, 2023-07-09:
PR: https://github.com/olliw42/mLRS-docu/pull/51

olliw42, 2023-07-09:
great!! Thx!

I guess we can close this, but I think we want to keep the RSSI topic in mind to eventually, in some future, work it out.

jlpoltrack, 2023-07-09:
> I guess we can close this, but I think we want to keep the RSSI topic in mind to eventually, in some future, work it out.

Agreed, I will create a new issue for a 'Link Statistics Page' and assign myself.

closed 2023-07-09 by jlpoltrack