olliw42/mLRS-docu #78: Upcoming - Relay Page

jlpoltrack, 2023-09-16:
Introduce a page documenting a relay setup.  Images kindly provided by Risto: https://discord.com/channels/1005096100572700794/1005096101239603232/1152525658640101456

olliw42, 2023-09-16:
many thx!

a note however: I'm not sure when and if relay mode would make it to main. You may know my (increasing) hesitancy to add features to the main code base which (1) I can't or won't continously maintain myself and which (2) are not routinely used by at least one user (who is also willing to report issues). For the relay mode, condition (1) equals to true, and it is not clear when and if condition (2) would become false. :)

olliw42, 2023-09-21:
I ponder over whether I shouldn't change my opinion in this specific case, The code additions/changes for the relay mode are not very dramatic, and it seems to me not so much can go really wrong here. I.e., I guess we might consider adding it, if at least someone else has tested and confirmed it to work. What do you think?

jlpoltrack, 2023-09-21:
I personally don't have much use for a relay setup but I could at least validate that it works as expected (have enough hardware lying around :) ).  I'll start a page and can just leave it as a branch if the relay stuff takes time to make it to main.

olliw42, 2023-09-22:
I myself also don't have any use of relay. :)

It's however the only feature I currently see which holds me back to do a v0.4 release. I mean, if it is ok it would be nice to include it, but I also think itÄ's time for a v0.4, so I would want to wait forever.

Maybe an approach could be to add it to the docs as an experimental feature, or whatever to call it. That way folks would know it exist and how to use it, and maybe would make someone use it. It could remain in the PR/branch, and we could go on with v0.4.

jlpoltrack, 2023-09-22:
PR is here: https://github.com/olliw42/mLRS-docu/pull/85

olliw42, 2023-09-22:
THX

so I assume you consider the outlined path to not postpone release of v0.4 because of relay reasonable, right?

jlpoltrack, 2023-09-22:
> so I assume you consider the outlined path to not postpone release of v0.4 because of relay reasonable, right?

No concerns from me

closed 2023-09-22 by jlpoltrack