olliw42/mLRS-docu #79: track some

olliw42, 2023-09-20:
docs went out of sync a bit, and may also see some additions
list may grow/shrink
some are just ideas

out of sync:
- [ ] https://github.com/olliw42/mLRS-docu/issues/76
- [ ] Model/Receiver/Config Id, how it works how to use

ideas:
- [ ] list firmware limitations on device pages, e.g. FRM303 won't offer MavlinkX, FLRC
- [ ] easy-to-solder devices page ?
- [ ] wireless bridge page ?
- [ ] OLED page ?
- [ ] FRM303, comments on power options ?

olliw42, 2023-09-21:
@jlpoltrack just to confirm, you don't see any reasons why FLRC shouldn't be included in main, right? I.e. works for you?

jlpoltrack, 2023-09-21:
> @jlpoltrack just to confirm, you don't see any reasons why FLRC shouldn't be included in main, right? I.e. works for you?

No concerns from me.

olliw42, 2023-09-21:
great. So we'll make it permanent :) THX!!

jlpoltrack, 2023-10-05:
@olliw42  @rotorman 
Could you give some hints on how this PR works if we want to document generically? https://github.com/olliw42/mLRS/pull/111

I went through the diff and I see this line added twice:

value="com.st.stm32cube.ide.mcu.gnu.managedbuild.tool.c.compiler.option.optimization.level.value.os" valueType="enumerated"

But don't really understand what this does.  For example, would one only need to add these lines to any .cproject file to enable debug?  Thanks

olliw42, 2023-10-05:
I think the procedure is NOT to directly fumble in this file, but to use the IDE's dialogs to enter the information. This might not be accurate, but like steps in chapters III & IV & A1 here https://github.com/olliw42/mLRS/blob/main/mLRS/CREATE_TARGET_INSTRUCTIONS.md#iii-create-target-ide-config, just for Debug

olliw42, 2023-10-05:
I guess what one needs to do is

10. click Debug
12. follow exactly
13. follow exactly
16. follow exactly

if usb is needed

A1.2 follow exactly

you guys may want to check

jlpoltrack, 2023-10-05:
Would it make more sense to add the debug steps to that page itself?  Then we could just link to it from the Docs / Custom Targets Page?

olliw42, 2023-10-05:
well, it would make the install more complicated than it is already
'
I think not everyone will want debug, I actually think only few will.

rotorman, 2023-10-06:
> @olliw42 @rotorman Could you give some hints on how this PR works if we want to document generically? 

Indeed, I would also suggest to use STM32CubeIDE dialogs for this. Basically, presently the release configuration has all the necessary include paths correctly set, whereas some have not yet been added into debug configuration. The task to get debug going is matching/adding the missing parts (e.g. by just comparing the dialogs and copy/pasting the missing ones).

olliw42, 2023-10-06:
I just tested it for a rx and a tx target, and can confirm that, when a Relase target was created, the steps

10. click Debug
12.  follow exactly
13.  follow exactly
16.  follow exactly

and if usb is needed

A1.2 follow exactly

make it to compile. If it is working I can't tell, but it compiles succesfully.

There are few vtable erros currently, which I will correct ASAP.

jlpoltrack, 2023-10-20:
@olliw42 Would image like this be sufficient for initial flashing? I thought about adding in ST-Link hardware but there are many variations and I think wouldn't want to recommend the cheap ones.

[image]

olliw42, 2023-10-20:
I would leave that totally you. I guess I would add a cheapo STLink just for the sake of teh picture looking more complete, but I think a diagram like yours is equally useful. Pl decide according to your taste :)

olliw42, 2023-10-24:
all done I guess :) so closing

closed 2023-10-24 by olliw42