olliw42/mLRS-docu #116: Wireless Bridge Mission Planner connection help.

ArtimisFowl888, 2024-01-09:
I am trying to use the M5Stamp C3U Mate conected to an R9M for the wireless bridge mode. 
when I try to connect to the autopilot in mission planer, I get stuck at the mavlink connecting... dialogue.

Is there a param or mLRS setting that I might have missed?

ArtimisFowl888, 2024-01-09:
I rebooted my computer and am not longer able to replicate the mavlink connecting... dialogue.
instead, I just get connection failed.

jlpoltrack, 2024-01-09:
Can you share the Arduino sketch you uploaded?  Did you uncomment the board define?

ArtimisFowl888, 2024-01-09:
[mlrs_wireless_bridge_copy_OAIRE.zip](https://github.com/olliw42/mLRS-docu/files/13876804/mlrs_wireless_bridge_copy_OAIRE.zip)

Here is the sketch I am using.

jlpoltrack, 2024-01-09:
A few thoughts:

- The attached has no versions uncommented, 'MODULE_M5STAMP_C3U_MATE_FOR_FRSKY_R9M' is the one you want because it accounts for the R9M inverted serial
- The baud rate on the Tx module (can be set via Lua) should be 115200, this is the default so not sure if you've changed

ArtimisFowl888, 2024-01-09:
[mlrs_wireless_bridge_copy_OAIRE2.zip](https://github.com/olliw42/mLRS-docu/files/13877010/mlrs_wireless_bridge_copy_OAIRE2.zip)

Yes, sorry I just relized the to .ino file was set to read only for some reason and my edits had not been saved. I have unlocked the file and reuploaded the sketch. 

I have not changed any baud rates other than the RX and the Serial1_baud to 38400.

jlpoltrack, 2024-01-09:
@brad112358 Do you have any thoughts here?  Figured that using the C3U Mate should be pretty straightforward.

olliw42, 2024-01-09:
@ArtimisFowl888 we will not kick you out here but pl consider the next time to use one of the discussion&support channels, i.e. the discord channel or the rcgroups thread. The github issues are generally NOT the place for support and trouble shooting. THX.

on your matter, can you pl post a picture of a setup

ArtimisFowl888, 2024-01-09:
MY apologies I will use the Discord next time.

I have a R9MM connected to a MATEK F405-WING-V2 on UART1 and SBUS in pin.
For the TX I am using a R9M with the M5Stamp C3U.

[image]
[image]

olliw42, 2024-01-09:
many thx. the intresting bit cannot be seen, namely how the wiring from the esp32 to the r9M board looks like :). The ESP32 does hve the labeling "C3U" on the bottom side.

jlpoltrack, 2024-01-09:
@ArtimisFowl888  Doesn't the Dip switch need to be up for Serial?  From the Docs:
[image]

brad112358, 2024-01-09:
Yes,  also what is the led on the esp32 module doing?

ArtimisFowl888, 2024-01-09:
> @ArtimisFowl888 Doesn't dip switch 1 need to be up for Serial? From the Docs: ![image](https://private-user-images.githubusercontent.com/41841496/295319886-ed45bdda-90c9-4fe0-bfb1-760507ead361.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4Mjk4NzksIm5iZiI6MTcwNDgyOTU3OSwicGF0aCI6Ii80MTg0MTQ5Ni8yOTUzMTk4ODYtZWQ0NWJkZGEtOTBjOS00ZmUwLWJmYjEtNzYwNTA3ZWFkMzYxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA5VDE5NDYxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI3YTM5ODFmYmI5ZWY1MTI2ZGViMzZlZTFmOGJkMDVjMjk5OWE0Yjg0MzhlMWJkYWFlMDIxODJmYTY4MWUxZWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.UkOTzJIJFQNcyOqUW0d-2_Zkchu0z1bbsf5ZU9K0adU)

That was the issue. I remember reading it when I first set it up, I must have missed it when trouble shooting.

Thank you, for all the help and keep up the great work.

jlpoltrack, 2024-01-09:
Good to hear - I will close this issue then as all is working.

closed 2024-01-09 by jlpoltrack