olliw42/mLRS-docu #171: CLI mode on ELRS rx

Rothammel, 2024-06-13:
According to the doc, I should set these parameters via cli:
"Set the following parameters via the CLI or Lua script:

Rx Out Mode = crsf
Rx Ser Baudrate (must match the baudrate for which the flight controller's MAVLink serial port is configured):
57600 for 31 Hz, 50 Hz
38400 for 19 Hz (or 57600, see SERIALx_BAUD)
230400 for FLRC
Rx Ser Link Mode = mavlink
Rx Snd RadioStat= ardu_1"

How can I do this? How can I activate CLI?

jlpoltrack, 2024-06-13:
> How can I do this? How can I activate CLI?

As it depends on what you're using, the hardware specific pages should give you some hints.

PS - The Discord (https://discord.com/invite/vwjzCD6ws5) is the best place for troubleshooting :)

olliw42, 2024-06-13:
@Rothammel 

the receiver is configured via the transmitter module.
That is, you need a tx module, bounded and connected to your receiver, and then access the cli of the transmitter module.
The receiver itself has no way to be configured.

as JLP suggested, these issues here are not for help and support questions. You should consider going to the discord or rcgroups places for help and support.

:)

closed 2024-06-13 by olliw42