olliw42/mLRS-docu #198: Check CRSF Elements on BW Radio

jlpoltrack, 2024-09-19:
Other elements may show up on BW Radio (such as VFAS) - do a sanity check and update the CRSF sensors page accordingly.

olliw42, 2024-09-20:
this VFAS thing is really surprising to me
it's just simply not in teh EdgeTx/OpenTx code
EdgeTX: https://github.com/EdgeTX/edgetx/blob/main/radio/src/telemetry/crossfire.cpp#L28-L59, https://github.com/EdgeTX/edgetx/blob/main/radio/src/translations/en.h

I wonder if this is an old firmware thing

jlpoltrack, 2024-09-20:
I took a quick look and my X9D does show that element for some reason on latest EdgeTx.  VFAS is a specific FrSky thing - could the passthrough stuff cause it? https://github.com/EdgeTX/edgetx/blob/edb04cc83373f9f13dce8ad69e671343883dd75c/radio/src/telemetry/frsky_sport.cpp#L61

olliw42, 2024-09-20:
our passthru handling can't do it since it also doesn't do have VFAS sensor
this could however be a yappu script thing, I believe to recall to have seen in it's code the option to set some radio sensors, maybe it's config in the yappu

closed 2025-01-03 by jlpoltrack