olliw42/mLRS-docu #216: R9 Bootloader Links No Longer Work

jlpoltrack, 2024-10-27:
With ELRS deprecating STM32 support the FrSky R9 ELRS bootloader links need updating.

Can use 3.x.x maintenance branch, e.g. https://github.com/ExpressLRS/ExpressLRS/blob/3.x.x-maintenance/src/bootloader/

brad112358, 2024-10-29:
Perhaps we should make our own copy in preparation for the day when 3.x is also deprecated in case they delete that branch?

jlpoltrack, 2024-10-29:
I will fix links in short term and then that's probably the better solution.

jlpoltrack, 2025-01-03:
@brad112358 I think we'd also want to have a copy of the https://github.com/ExpressLRS/ExpressLRS/tree/3.x.x-maintenance/src/python folder which includes the necessary files to flash.

Anyone know if is it acceptable to redistribute the files in mLRS docs?

olliw42, 2025-01-03:
I guess one should better ask them.

technically, I think, we don't have too. Since it's OS with a certain licence, we can fork it, or copy it, for as long as we correctly give the source and include the licence. Or whatever the license says.

brad112358, 2025-01-03:
I guess we can ask to be polite, but, since it's all licensed under GPL V3, the answer has to be yes as long as license it under GPL V3.

jlpoltrack, 2025-01-03:
So am thinking to have an archive in the files folder with:
- bootloader folder
- python folder
- GPL3 License (see below)

> correctly give the source and include the licence

How should this be done?

For the license I can include the GPL3 License file from their repo: https://github.com/ExpressLRS/ExpressLRS/blob/master/LICENSE but am unsure how/where to give the source, can this just be a text file?

olliw42, 2025-01-03:
yo, I think just including a README.md with some text and stating the licence would do it. One could also include the LICENCE.md, but don't think that's mandatory since it's just GPL3, but since it doesn't hurt to do I would do.

olliw42, 2025-07-17:
@jlpoltrack @brad112358 isn't this essentially resolved?

i.e., can we close?

closed 2025-07-18 by olliw42