olliw42/mLRS-docu #226: Lua script location needs to be given correctly

olliw42, 2024-11-27:
title says it

jlpoltrack, 2024-11-28:
This looks snippet from the Lua page looks to me - can you provide more specifics?

'The Lua script "mLRS.lua" or "mLRS-bw.lua" located in the "lua" folder of the mLRS github repository needs to be copied to the "SCRIPTS/TOOLS" folder of the radio's SD card. One can follow the common tutorials for how to do this.'

olliw42, 2024-11-28:
not really ok. The lua changes - not often but every once in a while - which can make it incompatible with previous versions. I.e., one cannot in general assume that the lua in main works with (all) previous firmware versions.
So it is with the lua in the main, it specifically asks for a minimum version of 1.3.03 - so will complain when used with e.g. v1.3.02. A user may however have difficulties to find currently the lua she/he should use ...

jlpoltrack, 2024-11-28:
Ok - so links for specific versions of the Lua and the corresponding mLRS versions they support is what is wanted?

olliw42, 2024-11-28:
I'm not totally sure how to handle that at this point in time.

so far we have proper tags only for stable versions, like 1.3.00. I want to change this to also have proper tags for the pre-release versions. So, v1.3.04 will be properly taged, so we can refer usesr to it. But we haven't done so far, so so far it's really ugly. I hope that we may not be that far from a v1.4.04, so maybe we leave it until when. The issue is just to note taht we need to give the users a better chance.

Maybe as quick fix we provide a link for v1.3.00 and v1.3.02.

olliw42, 2024-12-11:
we do hve now a proper link for v1.3.04 lua scripts https://github.com/olliw42/mLRS/tree/v1.3.04-pre-release/lua

closed 2024-12-11 by jlpoltrack