olliw42/mLRS-docu #250: Adaption needed for upcoming v1.3.06

olliw42, 2025-03-22:
- [ ] Update to reflect "19 Hz" vs "19 Hz 7x" modes for sx126x vs sx127x (probably needs many places to change)

jlpoltrack, 2025-03-22:
Second bullet is covered on the ELRS Tx Modules page (not sure why i didn't close this out).  https://github.com/olliw42/mLRS-docu/blob/master/docs/ELRS_TX_MODULES.md

jlpoltrack, 2025-03-26:
@olliw42  Could you provide new image with schottky removed for E77 MBL; https://github.com/olliw42/mLRS-docu/blob/master/docs/EBYTE_E77_MBL.md#as-tx-module ?

I've removed for E5 and EasySolder never mentioned.

olliw42, 2025-03-26:
@jlpoltrack modified the picture and removed the Schottkly diode comment

jlpoltrack, 2025-04-05:
Re: Flashing, some quick thoughts:

- Should add the Desktop App to the main repo's ReadMe Flashing section
- Will replace existing flashing on hardware pages with Desktop App specific ones

olliw42, 2025-07-17:
looking over it I think that while not all boxes are checked we have done the relevant ones, and the open ones are rather optional improvments.

so am closing. If anyone of you guys feels this is premature, please reopen :)